wellenvogel/avnav #7: when tapping an AIS target on Ipad safari the AIS page is only shown shortly

wellenvogel, 2015-07-19:
reported by Christian:
Auf dem Ipad unter Safari gibt es folgendes Problem:
Tippe ich auf der Karte auf ein AIS Ziel, so wird kurz die Tabelle der AIS Ziele angezeigt und dann sofort wieder die Karte. Manchmal springt er aber auf der Karte zu einem ganz anderen AIS Ziel und wechselt ggf. auch den Kartenausschnitt dahin.
Gleiches Verhalten habe ich auf dem Ipad auch mit den Browsern iCab und Mercury.

wellenvogel, 2015-07-19:
Unable to reproduce this on an Ipad 3, IOS 8.3

sixten828, 2015-09-19:
had similar issue on my Samsung S3 mini (with version 0.2.0)
havn't recognized it on my Samsung Tab 3

wellenvogel, 2017-02-13:
closed with c5b6157a86a52178df1aeeb1bdebbb2bf3019f50
after the complete restructure of the GUI this should be closed

closed 2017-02-13 by wellenvogel