wellenvogel/avnav #9: Navigate to a waypoint out of a list of waypoints

sixten828, 2015-09-19:
Route navigation is already implemented and seems to be quite good.
Also defining waypoints for a route is implemented.
So there should be already a management of a waypoint list implemented.

1. I would request the ability to read in a waypoint list from a file (which is already there) but I want to generate these waypoints with another tool a in a well known format.

2. Regularly I do not use route navigating but navigating to a waypoint.
Navigating to a marker is already implemented this needs to be enhanced to choose a "marker" from the existing waypoint list.

Above mentioned enhancements doesn't necessarily need more menues or buttons to keep the UI simple.

Thanks for this nice piece of navigational app I really appreciate it :-)

sixten828, 2015-10-10:
From 1.10.2015 onwards the Kieler Förde has a huge restricted (temporary) area also in the main shipping channel which is not in the electronic charts.
It is hard to know where the resttricted area is unless you have it visible on your chartplotter.

Having the ability to add remaining/loadable waypoints would help to visualize the restricted area.

wellenvogel, 2015-11-04:
Fore sure an interesting Funktion.
Potentielle a 2 step approach:
1. Option for a gpx overlay that will show symbols in the map ( no interaction)
2. real waypoint list with Selection Option.

free-x, 2017-09-21:
Hi, 

here for example a list of WayPoints from NV 1,2,3,4,5. They are always marked in PaperCharts.

[Wegepunkte2016.gpx.txt](https://github.com/wellenvogel/avnav/files/1320859/Wegepunkte2016.gpx.txt)

wellenvogel, 2020-12-19:
implemented with [20201219](https://github.com/wellenvogel/avnav/releases/tag/release-20201219)

closed 2020-12-19 by wellenvogel