wellenvogel/avnav #13: More NMEA data

wellenvogel, 2016-04-24:
We should be able to parse and display more nmea data. On the server side this should be handled by some pluggable architecture to be able to add new parsers on the fly. On the viewer we should have a widget approach with the ability to select the widget to be displayed at various positions (nav page, gps page). Potentially more gps data pages. There should be some options to easily add more standard widgets (i.e. no dedicated graphical stuff but simple value formatting).
- [ ] provide a plugin infrastructure on server (read all <prefix>_<name>.py files in server and user dir, init them). They should contain a class with some init function returning the nmea sentences they are interested in (optionally none). The class must also have a handleNmeaData function that will parse the data. On init it will receive a NavData interface with 2 functions: storeNavData (object similar to what we have) and storeNMEAData to write out some NMEA data. Optional: configuration
- [ ] restructure the NavData (server) see #12 
- [ ] pass NMEA data to plugins (instead of gpsd pased on plugin patterns)
- [ ] provide some plugins for basic data (RMC,...)
- [ ] build a widget factory on the client side
- [ ] base the display of data on widgets

free-x, 2017-01-14:
Hi Andreas,

i'm using sometimes for personal purposing pynmea2 library (https://github.com/Knio/pynmea2)

For example:
`#!/usr/bin/env python

import sys
import pynmea2

if len(sys.argv) < 2:
   f=sys.stdin
else:
   f = open(sys.argv[1])

for line in f:
  if line.startswith("!"):
    continue
  elif line.startswith("$"):
    msg = pynmea2.parse(line)
    if type(msg) is pynmea2.types.talker.RMC:
      print "RMC %s%s %s%s sog=%s cog=%s" % ( msg.lat,msg.lat_dir, msg.lon, msg.lon_dir, msg.spd_over_grnd, msg.true_course)
    elif type(msg) is pynmea2.types.talker.GGA:
      print "GPSFix at %s%s %s%s. Sat %s in use %d quality" % ( msg.lat,msg.lat_dir, msg.lon, msg.lon_dir, msg.num_sats, msg.gps_qual)
    elif type(msg) is pynmea2.types.talker.GLL:
      print "ValidStatusFix %s%s %s%s" % ( msg.lat,msg.lat_dir, msg.lon, msg.lon_dir)
    elif type(msg) is pynmea2.types.talker.VTG:
      print "SOG %f" % msg.spd_over_grnd_kts
    elif type(msg) is pynmea2.types.talker.DBT:
      print "Depth: %s" % msg.depth_meters
    elif type(msg) is pynmea2.types.talker.MWV:
      print "Wind: %s %s %s%s" % ( msg.reference, msg.wind_angle, msg.wind_speed, msg.wind_speed_units)
`
Would that way be helpful?

wellenvogel, 2017-01-14:
Looks promising. Thanks, I will try it out. Would like to go away from gpsd anyway...

wellenvogel, 2019-12-24:
With 20191223 the infrastrcuture is there. The plugin interface provides the functions. Documentation is still ongoing.

closed 2019-12-24 by wellenvogel