wellenvogel/avnav #20: Integrate an umts/lte stick

wellenvogel, 2016-12-20:
It would be nice to have an integration for such a stick including the option to set the pin.

free-x, 2017-01-14:
Hi Andreas,

UMTS/LTE stick is not more modern. I'm proposing another way: USB Tethering. Dual SIM Smartphones are not more rarely. 
If in Smartphone is USB Tethering configured, you can connect Raspberry over USB-Data Cable and Raspberry gets IP on USB0 Interface. Small changes for MASQUARADING over USB0 also needed.

wellenvogel, 2017-01-14:
Nice option too. But you can use the normal hotspot function of a mobile if you have a second WLAN adapter. This also gives you the option to centrally connect to a port WLAN...

wellenvogel, 2021-03-26:
Currently no plans to go any further here.

closed 2021-03-26 by wellenvogel