wellenvogel/avnav #23: Make overlays independet from map

wellenvogel, 2016-12-20:
To be able to replace the map layer, the overlays for nav, track, route, ais should be independent on a separate canvas.

W-Geronius, 2020-11-06:
Detaching overlays from map would also permit implementing enhancement issue #9. 

> Potentially a 2 step approach:
> Option for a gpx overlay that will show symbols in the map ( no interaction)
 
 Another example (extract) as used by nautin.nl to provide up to date navigation marks as an overlay in OpenCPN, where field <sym> refers to externally defined icons
 
 ```
 <?xml version="1.0" encoding="ISO-8859-1" standalone="yes" ?>
 <gpx
  version="1.1"
  creator="Nopoint - http://www.nautin.nl/"
  xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
  xmlns="http://www.topografix.com/GPX/1/1"
  xsi:schemaLocation="http://www.topografix.com/GPX/1/1 http://www.topografix.com/GPX/1/1/gpx.xsd">
 <metadata>
 <name>Alles.gpx</name>
 <time>2019-05-06T11:01:12Z</time>
 <keywords>1234670</keywords>
 </metadata>
 <wpt lat="51.68836666666665" lon="4.54565">
 <name>1</name>
 <sym>227</sym>
 <cmt>geel</cmt>
 <desc>geel</desc>
 <extensions><opencpn:scale_min_max UseScale="true" ScaleMin="200000" ScaleMax="0" /></extensions>
 </wpt>
 <wpt lat="51.434749999999994" lon="3.636483333333334">
 <name>1</name>
 <sym>014</sym>
 <cmt>groen</cmt>
 <desc>groen</desc>
 <extensions><opencpn:scale_min_max UseScale="true" ScaleMin="200000" ScaleMax="0" /></extensions>
 </wpt>
</gpx>
```

wellenvogel, 2020-11-06:
What you address here is something different.
Anyway this has been on the list for long time...
I will follow this with #77.

closed 2020-12-19 by wellenvogel