wellenvogel/avnav #35: route can not be activated if default.gpx does not exist

free-x, 2017-09-21:
Hallo Andreas,

hier ist ein Bug: wenn in avnav/data/routes kein default.gpx existiert, die Route, die über Web oder SSH hochgebeamt wurde, in Routen-manager ist versteckt und nicht aktivierbar. In Download-Manager ist aber sichtbar und kann gelöscht werden.

Regards
free-x

wellenvogel, 2017-09-21:
Today routes will only become visible if they are uploaded via avnav. If you copy them into the routes dir by hand, they will only become visible after a restart of avnav.
I'm wondering what exactly you did - basically I tested this and the route was not visible both in the routes editor and on the download page.

benvanstaveren, 2020-08-27:
I actually ran into this same issue; even re-creating an empty (0 waypoint) default.gpx in the server's routes directory did not solve the issue. However, the issue seems to be in the React app; when I start off with a blank default route in the route editor, I click the little route panel to get the list of routes, then click the (empty) default route to save it under a new name (e.g. testroute), it will overwrite the existing default one and will just rename it instead.

However, on the server end, you do see the "new" route show up as a gpx file. The server now has a default.gpx, and a testroute.gpx - the local store in my browser also shows an avnav.route.default and avnav.route.testroute key - however, the route list in the route editor will still only show testroute - in fact, closing out the route editor and going back into it will now only show the default route. Uploaded routes are never visible.

wellenvogel, 2020-08-28:
Thanks for reporting. Will check this again.
What do you mean with  " Uploaded routes are never visible"?

wellenvogel, 2020-11-02:
Unable to reproduce this with the latest version (20201102).

closed 2020-11-02 by wellenvogel