wellenvogel/avnav #41: [enhancement] UDPwriter/broadcaster

free-x, 2019-03-19:
Hi Andreas,
I started implementing this extension. You can see the progress in my repository
https://github.com/free-x/avnav/tree/udpwriter
It's not working yet. I'm having problems instantiating. If you have idea, would be very grateful for any tip

Regards
free-x

wellenvogel, 2019-03-19:
Thx!
Works great.
Used the config:
`<AVNUdpWriter address='localhost' port="9999"/>`
And tested with `nc -l -k -u localhost 9999`.
I would suggest renaming address to host as this is used in other writers.
And would maybe use localhost as default destination. And maybe also have a default port...
And I guess we can omit the inheritance from SocketReader and setting "self.feederWrite"
Should I just "hijack" the file or would you submit a pull (would be better as you will be visible as the author ;-))?
Regards
Andreas

free-x, 2019-03-20:
Hi Andreas,

I'm gonna clean up some code tonight and test it again. Only then will I open PR. Stay tuned

Regards
free-x

free-x, 2019-03-20:
I have tested with Navionics....Seems to be working. My Config
```<AVNUdpWriter host="192.168.30.255" port="2000" filter=""/>```

Result:
[image]

free-x, 2019-04-13:
Merged

closed 2019-04-13 by free-x