wellenvogel/avnav #43: NMEA2000 worker/feeder

free-x, 2019-04-04:
Hallo @wellenvogel ,

wie schon früher angedeutet, habe ich diesen Winter mich mit N2K beschäftigt. Quick&Dirty läuft das schon.
Habe für experimente Waveshare [CAN Hat](https://www.waveshare.com/rs485-can-hat.htm) benutzt. 

1. CAN Hat aktivieren in /boot/config.txt
```
dtparam=spi=on
dtoverlay=mcp2515-can0,oscillator=8000000,interrupt=25,spimaxfrequency=1000000
```
2. Netzwerk Interface /etc/network/interfaces.d/can0
```
#physical can interfaces
allow-hotplug can0
iface can0 can static
bitrate 250000
down /sbin/ip link set $IFACE down
up /sbin/ifconfig $IFACE txqueuelen 10000
```
3. Für Kommunikation werden can-utils und canboat benötigt. Mein canboat kompilat mit erweiterten features wie AIS im Anhang
```
sudo aptitude install can-utils
sudo tar xzvf canboat.tar.gz -C /
```
4. Das ganze Kette ans Start bringen (/etc/rc.local)
```
candump can0 | candump2analyzer | analyzer -json | n2kd
```
5. Dem AvNav abholen beibringen (avnav/data/avnav_server.xml)
```
<AVNSocketReader  host="127.0.0.1" port="2599" />
```
That's all.
Falls man als Zubringer irgendwelche Actisense, YachtDevices in raw modus benutzt. 
```
actisense-serial -s 230400 -r /dev/ttyUSB0 | analyzer -json | n2kd
```
Jetzt kommt die Frage ;-)
Soll das als Handler laufen? Oder lieber so lassen? 
Ideen? Kommentare?
Gruß
free-x
[canboat.tar.gz](https://github.com/wellenvogel/avnav/files/3044504/canboat.tar.gz)

wellenvogel, 2019-04-05:
Hi @free-x ,
danke Dir für die tollen Beiträge.
Mein Vorschlag wäre:
Eigenes Paket daraus bauen (vielleicht direkt mit einem service drin) - dann kann man das auch leichter separat pflegen. Toll wäre so mit Vollkomfort - also einfach Paket installieren und dann ist (fast) alles fertig.
Man muss ja dann nur bei avnav den socket reader konfigurieren - können wir dann in das template noch auskommentiert mit aufnehmen.
Sonst wäre der neue Weg: Plugin schreiben (ist etwas simpler als ein Handler - und kann auch in ein separates Paket). Da fehlt aber noch ein wenig Doku dazu.

Ich kann es selbst ohnehin nicht testen - auch deshalb wäre mir ein separates Paket lieber.
Gruß Andreas

free-x, 2019-09-06:
Hi @wellenvogel ,

habe ein debian package für Debian Buster am start
Download : https://www.free-x.de/debian/pool/main/c/canboat/canboat_1.2.1-20190831_armhf.deb

Gruß
free-x

wellenvogel, 2022-01-14:
I think with the current set up with canboat being preconfigured and the AvNav n2k plugin we should have addressed this sufficiently.

closed 2022-01-14 by wellenvogel