wellenvogel/avnav #57: convertercrash

Hartholz, 2020-06-06:
Hey Andreas,
Ich habe über das Programm Sat2Chart Navionicskarten aus "SASPlanet" im mbtiles-Format für OpenCPN bezogen. Bei der Konvertierung der Karten in ein AVNAV-Kompatibles Format ergibt sich folgende Fehlermeldung:
2020/06/06-17:54:43 ERROR: Traceback (most recent call last):
  File "C:\Program Files (x86)\AvNavNet\chartconvert\read_charts.py", line 1786, in <module>
    main(sys.argv)
  File "C:\Program Files (x86)\AvNavNet\chartconvert\read_charts.py", line 1725, in main
    sys.exit(1)
SystemExit: 1

ERROR: Traceback (most recent call last):
  File "C:\Program Files (x86)\AvNavNet\chartconvert\read_charts.py", line 1786, in <module>
    main(sys.argv)
  File "C:\Program Files (x86)\AvNavNet\chartconvert\read_charts.py", line 1725, in main
    sys.exit(1)
SystemExit: 1
***Converter finished ***
2020/06/06-17:54:43 ERROR: conversion failed

Die mbtiles-Datei verfügt über zwei Zoomstufen!
Danke für die Arbeit!
Grüße!
-Hartholz

free-x, 2020-06-06:
Hi @Hartholz ,

seit version 20200325 konvertierung von mbtiles ist nicht mehr nötig. AvNav versteht mbtiles dieses format "out-of-box". Die Datei einfach ins avnav/data/charts/ kopieren.

Handbreit
free-x

free-x, 2020-06-06:
eigentlich fehler deutet darauf hin, dass das Verzeichnis nicht existiert

Hartholz, 2020-06-06:
Es handelt sich um die Android App. Beim normalen Laden der Mbtiles-Karte über die Uploadfunktion streikt Avnav nach dem Öffnen der Karte. Lediglich ein roter Rahmen erstreckt sich über die halbe Welt.

free-x, 2020-06-06:
AvNav startet in zoom 10. Hast du probiert weiter reinzu"zoom"en?
Was für Zooms beinhaltet die Datei?
Habe gute erfahrungen mit 10,12,14,16 (in SAS Welt ist das 11,13,15,17 Stufen) gemacht

Hartholz, 2020-06-06:
Die mbtils wurden in den Zoomstufen 17 bis 19 erstellt.

free-x, 2020-06-06:
wie gesagt, probier mal reinzoomen. Am linken Rand siehst du die aktuelle Zoomstufe

Hartholz, 2020-06-06:
[image]
Der Rote Punkt ist die geladene Karte und sollte eigentlich eine Bucht in Kroatien in der Zoomstufe 17 und 19 darstellen.

free-x, 2020-06-06:
du bist in Zoomstufe 4. Und bis 17 gibt's keine Tiles. Zoom rein, Luke!

Hartholz, 2020-06-06:
°,° Stimmt! Trotzdem zieht sich bei der Auswahl von "xyz" ein roter Kasten von Kroatien aus über ganz Afrika. Ist das so richtig oder handelt es sich trotz meiner Blindheit um einen Bug?

wellenvogel, 2020-06-06:
Könntest Du die Datei mal bereitstellen?

Hartholz, 2020-06-06:
https://send.firefox.com/download/843bc634175783fe/#f-6yie0-KskO9tNfeGYNnA

free-x, 2020-06-07:
"zieht sich bei der Auswahl von "**xyz**" ein roter Kasten von Kroatien aus über ganz Afrika

ist es aber nicht. anscheinend ist das "tms"
```
sqlite> select * from metadata;
name|2020 Mitte Z17+Z19 20200221 Meters
type|baselayer
version|1.2
description|Created by SAS.Planet
format|png
bounds|14.23278809,44.17432484,15.22705078,44.61784415
attribution|
scheme|tms
minzoom|16
maxzoom|18
center|14.72991943, 44.39608450, 16
```

metzger100, 2020-06-07:
Ich habe den Fehler bei den SAT2Chart SASPlanet auch. Die Kroatienkarte befindet sich bei dem xyz-schema im antarktischen Pazifik.

wellenvogel, 2020-06-07:
So, habe jetzt noch mal ein wenig gespielt.
Wenn man die Datei auf "xyz" setzt, scheint es OK zu sein. Allerdings gibt es erst ab zoom level 16 was darin. Das ist potentiell schwer zu finden.
In der raspberry Version klappt das Umsetzen auf "xyz" leider gerade nicht, Korrektur kommt.
[image]
[image]
[image]

wellenvogel, 2020-06-07:
Noch ein Hinweis:
Unter android ging es spontan. Man muss aber schon genau auf level 16 oder 18 sein, wenn man erst mal was finden will.
Ich erzeuge meine Karten daher immer beginnend bei mindestens zoom 8 oder so. bSicher könnte man da nochmal was tun - aber im Moment ist es genau so.

free-x, 2020-06-07:
Ich glaube es ist ein Bug in SAS: Die speichern XYZ struktur und nennen das "tms" und umgekehrt.

wellenvogel, 2020-07-26:
Nach einiger Diskussion haben wir festgestellt, das xyz und tms in Avnav genau verkehrt herum genutzt werden.
Wenn man nichts einstellt, geht es im Normalfall, weil der default tms ist. AvNav nennt das (im Moment) zwar intern xyz, ntut aber das Korrekte.
Das wir in einer der nächsten Versionen korrigiert - man muss dann u.U. die Karten, bei denen man das explizit gesetzt hat, noch einmal umsetzen. Ich werde in den release Hinweisen noch einmal darauf aufmerksam machen.

Im Moment sollte es als für Karten im tms Format (default) ohne Probleme gehen. Wenn man Karten im xyz Format hat, muss man in AvNav im Moment explizit "tms" einstellen - bis zur Korrektur.

metzger100, 2020-08-04:
Sehr gut, Danke. Kannst du einen Beitrag auf deiner Website dazu machen, wie man mbtiles merged? Hab ne große Karte von Zoom 10 bis 17 erstellt und würde gerne ein paar Buchten in Zoom 18-20 einfügen. Ich habe das mit dem Konverter aber nicht hinbekommen. Das wäre einen super Sache.

wellenvogel, 2020-08-04:
Hmm,
Du brauchst für mbtiles keinen Konverter.
Einen merger für mbtiles hatte ich nicht spontan gefunden - der Konverter kann das auch nicht.
Ich habe sowas mal ( unabhängig von avnav) gemacht - 
https://github.com/wellenvogel/mbtilesmerge
Ist aber nur Kommandozeile. Vielleicht hilft Dir das ja weiter.

metzger100, 2020-08-04:
Okay. Das werde ich mal ausprobieren. Wäre cool wenn du den in den Konverter als Gui-Version einbauen könntest. Ich glaub nicht, dass alle mit Kommandozeilentools zurecht kommen und eine Mergefunktion von mehreren mbtiles wäre für Nutzer von Avnav schon ganz praktisch.

wellenvogel, 2020-09-05:
Correction for xyz/tms mbtiles handled with #63.

closed 2020-09-05 by wellenvogel