wellenvogel/avnav #59: dynamic adaption of course vector length according to boatspeed

doccodyblue, 2020-07-21:
It would be nice to see a dynamic adapted length of the course vector (and possibly the course vectors of AIS objects) according to ships speed. 

In other words: My ship will be 'here' (at the end of the course vector line) in 10 mins

So it would be easy to determine the estimated position in the future to aid navigation i.e. crossing high frequented traffic separation areas or to avoid collision in general

wellenvogel, 2020-07-26:
I will work on this. For the own ship this will be easy. For AIS symbols it could affect performance...

wellenvogel, 2020-09-05:
There is a new setting "Boat Course Vector Length(sec)" on the navigation tab. Here you can set the the number of seconds for the course vector (default 600s).
For AIS you can enable course verctors with the same length for AIS targets using "AIS Use Course Vector" on the AIS tab. If not set the old behavior is used (so if you encounter performance issues you can disable this setting).

wellenvogel, 2020-09-05:
First implementation with c8ed92ba66d014911bf6a524ac97583dd0ec7de8, daily [20200627](https://www.wellenvogel.net/software/avnav/downloads/daily//20200726/)

wellenvogel, 2020-11-05:
https://www.wellenvogel.net/software/avnav/downloads/release-ochartsplugin/

closed 2020-11-05 by wellenvogel