wellenvogel/avnav #60: parsing empty DPT or DBT throws exception

doccodyblue, 2020-07-21:
parsing empty DPT or DBT string throws an error. 

At least Standard Horizon and my Airmar DST 800 generate a valid NMEA0183 message without values when no valid measurement is possible (in case of too deep water or too shallow water)

This is more or less cosmetic, but spamming the log file (regarding IO operation on the SD card). Maybe that could be catched in some way.

Traceback (most recent call last):
  File "/usr/lib/avnav/server/avnav_nmea.py", line 341, in parseData
    rt['depthBelowTransducer'] = float(darray[1])
ValueError: could not convert string to float:

2020-07-21 21:10:21,735-615-[ 933]-GpsdFeeder[standalone feed]-INFO- error parsing nmea data $SDDBT,,f,,M,,F*28

doccodyblue, 2020-07-25:
This is also syntactically correct, I think. Coming from my DST800, 1.3 is in m. 

2020-07-25 07:54:31,738-594-[ 953]-GpsdFeeder[standalone feed]-INFO- error parsing nmea data $SDDPT,1.3,*7B

Traceback (most recent call last):
  File "/usr/lib/avnav/server/avnav_nmea.py", line 342, in parseData
    if float(darray[2]) >= 0:
ValueError: could not convert string to float:

free-x, 2020-07-25:
@wellenvogel 
I would propose to use something like this
```
def savefloat(f):
    try:
        return float(f)
    except:
        return 0.0
```

wellenvogel, 2020-07-26:
In general we can easily ignore such values. But what should be stored in this case? I guess just storing "0" values is not really appropriate. And if we ignore the values completely it could hide other errors - so it could be difficult to find out about problems (i.e. why I'm unable to see the values coming in?...). I will check if there is an easy way to only report the errors once (or seldom).

doccodyblue, 2020-07-26:
For the first case with the empty values it is not an error of AvNav so should it really show up in the log? 
In fact it is just another state of the transducer. So maybe it would be an option to set f to -1 and parse it out to let AvNav display "--", indicating to the user that depth is not available right now? That would represent the same behavior that the depth sounder shows to the user.

wellenvogel, 2020-07-27:
Yes, this sounds ok.
I will think a bit on it. I would like to see a difference between no data received and empty values - but maybe not that important.

free-x, 2020-07-28:
I think in this specific case, as a workaround, we can build in something like this
```
        try: 
            if float(darray[2]) >= 0:
               rt['depthBelowWaterline'] = float(darray[1]) + float(darray[2])
            else:
               rt['depthBelowKeel'] = float(darray[1]) + float(darray[2])
        except:
             pass
```

wellenvogel, 2020-08-10:
first correction with https://github.com/wellenvogel/avnav/commit/c35b3065d39404dd1c78660e66428c52ce9557a0

wellenvogel, 2020-11-05:
https://www.wellenvogel.net/software/avnav/downloads/release-ochartsplugin/

closed 2020-11-05 by wellenvogel