wellenvogel/avnav #63: correctly use xyz/tms format for mbtiles

wellenvogel, 2020-09-05:
as discussed in #57 currently AvNav inverts the meaning of xyz and tms when using mbtiles. This should be corrected.

free-x, 2020-10-28:
i'm confused now.
I have downloaded NOAA MBTiles
```
$ wget https://tileservice.charts.noaa.gov/mbtiles/50000_1/MBTILES_17.mbtiles
$ sqlite3 MBTILES_17.mbtiles
SQLite version 3.27.2 2019-02-25 16:06:06
Enter ".help" for usage hints.
sqlite> select * from metadata;
profile|global-mercator
attribution|NOAA
name|MBTILES_17
format|png
basename|MBTILES_17
bounds|-78.75, 39.9097362345, -73.125, 42.0329743324
minzoom|3
version|1
template|{{#__teaser__}} <div class='tooltip'> <strong>{{name}} {{kap}} </strong> <small class='tooltip'> NTM Date: {{ntm_date}} Scale: {{scale}} Units: {{units}} {{path}} </small> </div> <style type='text/css'> .tooltip strong { display:block; font-size:16px; } .tooltip small { font-size:10px; display:block; overflow:hidden; max-height:90px; line-height:15px; }</style> {{/__teaser__}}
maxzoom|19
tilejson|2.0.0
scheme|xyz
type|overlay
description|NEW YORK AND NEW JERSEY
```
And this is displayed correctly

wellenvogel, 2020-10-31:
Unfortunately there does not seem to be a real spec that describes the "scheme" entry in the metadata table.
And it seems that different sources sue it differently.
From the Spec (https://github.com/mapbox/mbtiles-spec/blob/master/1.3/spec.md):

`Note that in the TMS tiling scheme, the Y axis is reversed from the "XYZ" coordinate system commonly used in the URLs to request individual tiles, so the tile commonly referred to as 11/327/791 is inserted as zoom_level 11, tile_column 327, and tile_row 1256, since 1256 is 2^11 - 1 - 791.`

So TMS is the default format and uses inverted y coordinates.
Considering this the NOAA mbtiles seem to be wrong.

From my perspective we now have a couple of options:
1. make the change in AvNav and describe that e.g. for NOAA mbtiles you need to change to tms by hand. And you have to change the old files that you already have set once again.
2. completely ignore the scheme parameter and always assume tms. We still can provide an option for the user to switch to xyz and store this with a different name in the metadata table (just only be used by AvNav) - e.g. avnav_scheme.
This will have the advantage that old mbtiles in tms format (that follow the spec...) will continue to work.
3, use option 2. and ask the user if there are mbtiles files that have a scheme entry but have not been explicitly set by the user.

I personally would vote for option 3. I would just display the mbtiles files with a scheme entry and no avnav_scheme entry with a different color on both the entry page and the download page. So the user would know that maybe the scheme should be explicitly set.

free-x, 2020-10-31:
after all the mistakes we had experienced and based on the spec and experiences with MOBAC and SAS, which always store by default in TMS, I would tend to option 2. I think this will cover 99.9% cases and will work with all "old" charts

wellenvogel, 2020-11-05:
implemented wth [release-20201105](https://www.wellenvogel.net/software/avnav/downloads/release/20201105/fr)

closed 2020-11-05 by wellenvogel