wellenvogel/avnav #71: Compute True Wind

wellenvogel, 2020-09-05:
AvNav should be able to compute true wind data from received apparent and from cog/sog.
This datat should be available for the UI (including widgets) and as NMEA.
Preferrably it should auto detect wheter true wind data is available or not.

toolbar23, 2021-03-29:
SignalK has a plugin for calculating this kind of derived data that works quite well in my tests: https://github.com/SignalK/signalk-derived-data

Maybe it is less maintenance-work in the long rung to use/configure this?

W-Geronius, 2021-04-24:
@toolbar23 You are right, however including this in AvNav would also serve users who don't have SK installed (e.g. on their android device)

wellenvogel, 2021-05-21:
addressed now by  a plugin:
https://github.com/kdschmidt1/avnav-more-nmea-plugin

closed 2021-05-24 by wellenvogel