wellenvogel/avnav #72: Existing mbtiles set xml

Jamos1988, 2020-09-14:
Hi!

I'm trying to configure an xml to point to an existing local directory that includes mbtiles, but having trouble.
I use https://github.com/SignalK/charts-plugin so SignalK is able to use mbtiles in Signalk/Freeboard.

I realize that Avnav is capable of downloading mbtiles directly from OSM, but I'd rather use the ones I already have.
I've configured the following xml, but it doesn't work:

`<?xml version="1.0" encoding="UTF-8" ?>
 <TileMapService version="1.0.0" >
   <Title>OpenSemap/OpenStretmap</Title>
   <TileMaps>
     <TileMap 
       title="OpenSeaMap" 
       href="/home/pi/Charts/MBtiles/openstreetmap/openseamap/charts/mbtiles"
       minzoom="6"
       maxzoom="20"
       projection="EPSG:4326">
             <BoundingBox minlon="9" minlat="50" maxlon="20" maxlat="60" title="layer"/>
       <TileFormat width="256" height="256" mime-type="x-png" extension="png" />
    </TileMap>
    <TileMap 
       title="OSM" 
       srs="OSGEO:41001" 
       href="/home/pi/Charts/MBtiles/openstreetmap/openseamap/charts/mbtiles" 
       minzoom="4"
       maxzoom="20">
       <BoundingBox minlon="-20" minlat="10" maxlon="30" maxlat="70" title="layer"/>
       <TileFormat width="256" height="256" mime-type="x-png" extension="png" />
    </TileMap>
   </TileMaps>
 </TileMapService>

`

I know I have to change one of the href's, but not sure to what. 
I presume that I need to look at how https://github.com/SignalK/charts-plugin does this, but not sure where to look.

Hope you can  help!

free-x, 2020-09-14:
1) you can copy your MBTiles File to /home/pi/avnav/data/charts
2) OR change in avnav/data/avnav_server.xml in AVNHttpServer - Section
<Directory urlpath="maps" path="/path/to/your/mbtiles"></Directory>

wellenvogel, 2020-10-14:
I have some plans to use the signalk chart provider also for AvNav. When implemented it should work out of the box.

Jamos1988, 2020-10-14:
Hi, that sounds great! Really admire your work.

wellenvogel, 2020-11-05:
implemented wth [release-20201105](https://www.wellenvogel.net/software/avnav/downloads/release/20201105/fr)

closed 2020-11-05 by wellenvogel

Jamos1988, 2020-11-05:
Nice!!!