wellenvogel/avnav #79: prompt on closing avnav causes issues when using it with nativefier

mgrouch, 2020-12-11:
prompt on closing avnav causes issues when using it with nativefier.

Why trying to use it with nativefier in this project
https://github.com/bareboat-necessities/lysmarine_gen

Nativefied avnav application fails to close on clicking close button.
It's related to https://github.com/jiahaog/nativefier/issues/422

Can a confirmation prompt to close avnav be made optional (and possibly off by default)?

 Thanks,
--MG

wellenvogel, 2020-12-11:
Ok, I will think about...
Would an additional parameter like "....?noCloseDialog=true" be suitable for you?

wellenvogel, 2020-12-19:
implemented with https://github.com/wellenvogel/avnav/releases/tag/release-20201219

closed 2020-12-19 by wellenvogel