wellenvogel/avnav #80: gpg: no valid OpenPGP data found for avnav repo

mgrouch, 2020-12-19:
wget -q -O- https://open-mind.space/repo/open-mind.space.gpg.key | apt-key add - 

gives

gpg: no valid OpenPGP data found

free-x, 2020-12-19:
this repo is outdated. use https://www.free-x.de/debian/ 
```
deb https://www.free-x.de/debian buster main
```
GPG: https://www.free-x.de/debian/oss.boating.gpg.key

free-x, 2020-12-19:
you also have typo in your command
```
wget -q -O - https://open-mind.space/repo/open-mind.space.gpg.key | apt-key add -
```

pandel, 2020-12-19:
@free-x and @mgrouch : the repo is not outdated!

I made some bigger changes to my frontend server and had to take down the repo a couple of hours.

@free-x : why do you say it is outdated without contacting me first? The repo update algo is still running although I am not able to update avnavTouch at the momen...

wellenvogel, 2020-12-19:
@pandel: great to hear that the repo is still being updated.
There had been some issues in the forum where people had (potentially?) not been able to get the latest version on the touch image. So we assumed that it is not really updated.

free-x, 2020-12-19:
@pandel 
glad to see you again.

wellenvogel, 2020-12-19:
@pandel: can we have a short discussion in the Forum (PN is there...)?

wellenvogel, 2021-03-26:
nothing related to AvNav open here any more

closed 2021-03-26 by wellenvogel