wellenvogel/avnav #82: AIS Alarm

kdschmidt1, 2021-01-05:
i'm not able to get an AIS Alarm (assuming it should be activated as soon as the AIS Target is changing color to red).
i have inserted <Alarm command="sound" name="ais" parameter="$BASEDIR/../sounds/aisAlarm.mp3" repeat="1"/>
in the AlarmHandler section of avnav-server.xml but no sound and no alarm message on screen.
other alarms like mob or wp are working ok.

wellenvogel, 2021-01-05:
Currently there is no chance for an AIS alarm. All the ais computation is done on the client (Browser) - but the alarms need to be handled at the server (just consider that you potentially will not have any client active). So it would require to re-implement this computation at the server side.

kdschmidt1, 2021-01-09:
Using the signalk plugin one could use the 
signalk-derived-data app to activate a notification for configurable closest approach. i was able to catch the notification in the plugin, but i have no idea how to initiate the alarm out of the plugin. any idea?

wellenvogel, 2021-01-09:
Not possible (yet).

wellenvogel, 2022-04-03:
The next release will have an integration with the signalk notifications. So you can display and handle SignalK notifications in AvNav.
Current dailies already contain this function.

Fietsenwiel, 2022-06-16:
@[kdschmidt1](https://github.com/kdschmidt1), can you share the way you calculate the closed approach with the signalk-derived-data app?
This is a very welcome feature!

Thanks in advance

kdschmidt1, 2022-06-17:
In the Plugin-Config for the Derived-Data-Plugin you have to activate under "Traffic" the
Dangerous targets notification zone (CPA limit / TCPA limit => Notification level) with the "+"button and then edit the CPA and TCPA limit, select the Notification level and by checking the "Send notification for this zone" you will receive notifications from SignalK