wellenvogel/avnav #85: v20201227 Layout Editor: some bottom buttons not accessible for editing

W-Geronius, 2021-01-11:
editing bottom widgets of navpage is not possible, if buttons are automatically aligned in two rows on mobile phone or tablet. In edit mode bottom buttons will be displayed all in one row only, some moving out of accessible region.

It has not been checked whether or not this situation may happen with other screen layouts as well

[image]  [image]

wellenvogel, 2021-01-11:
For sure an issue.
Not that easy to fix due to the used libs for cut and paste. For now you would need to flip your device or use a bigger screen for editing the layout.

W-Geronius, 2021-01-12:
Flipping my on board 1024x768 pixel tablet will still leave some widgets left out.
At home a workaround would be to change layout on desktop computer

wellenvogel, 2021-01-12:
And you could decrease the widget fonts in the mean time as long as you are editing.

wellenvogel, 2024-09-22:
With the daily [20240914](https://www.wellenvogel.net/software/avnav/downloads/daily/20240914/) the widgets in should shrink to fit in the layout editor. Hopefully this will address this issue (although not looking great).

wellenvogel, 2025-07-23:
release with [20250723](https://www.wellenvogel.de/software/avnav/downloads/release/20250723/?lang=en)

closed 2025-07-23 by wellenvogel