wellenvogel/avnav #87: suggestion: WiFi connection lost alarm

W-Geronius, 2021-01-11:
Triggering an alarm at client side whenever client loses server connection might be a useful enhancement 
Imagine mobile phone being the only sound enabled device in state of anchor watch:
if Wifi connection fails -> anchor watch will fail without notice

W-Geronius, 2021-01-11:
This might be part of some dedicated alarm handling module, like for MOB, anchor watch, gps lost, waypoint reached or possibly other user definable alarms

wellenvogel, 2021-01-11:
yes, could be an option.
But in no case I would rely on a mobile device for the anchor alarm handling.
There is a lot of situations where the mobile OS could silently stop your browser without informing you. So you would have no chance for a sound.
I personally run a beeper at my pi that will handle the alarm sound.
See https://github.com/wellenvogel/avnav/tree/master/hardware/simple-pi

captainfast-git, 2021-01-20:
Maybe this can be feature for the Android version.

wellenvogel, 2021-01-20:
But for android it does not make sense either...
On the master device (even if in external browser mode) there is nearly no chance for connection lost (and the alarm is handled on the master even without browser connected).
And for another device - the same applies as for the pi use case.
But I guess we can have some sort of alarm for that any way (although not for the anchor alarm use case).

W-Geronius, 2021-04-24:
ok, ignore the anchor alarm use case. Another scenario, as it happened to me more than once: android tablet sitting in cockpit (as client to pi server) loses connection while your are sailing/motoring happily (possibly using detached autopilot). You don't watch the tablet, but are busy with something else, whatever - ok, bad seamanship. You certainly want to get notified if the client doesn't display valid data due to lost connection. Probably best (and easiest) to be implemented as layout option.

wellenvogel, 2021-11-09:
I guess this would not really fit this use case. This "wifi connection lost" is something at the client side (i.e. your browser device). And this device for sure has no way of sending/receiving NMEA data.

Fietsenwiel, 2021-11-09:
For wifi connectionlost I understand a mnea string is not relevant. But for Anchoralarm and MOB alarm it is.
Would be realy nice addition to this great software!

W-Geronius, 2022-04-05:
👍🏼

closed 2022-04-05 by W-Geronius