wellenvogel/avnav #88: suggestion: Moored ship and usage of HDM data

W-Geronius, 2021-01-11:
COG should be set to '---' when SOG is Zero and ship (arrow) aligned to HDM (if available) or changed to a solid circle.

While boat is moving, an option would be nice to display HDM and COG as separate vectors. 
HDM could then  determine the ship's (arrow's) direction.

wellenvogel, 2021-01-11:
Needs some further thoughts - see #95.
Doable at least.

Littlechay, 2021-05-24:
I support this request and as suggested by W-Geronius I would prefer HDT (Heading True) to be used.

W-Geronius, 2021-05-26:
@Littlechay is right of course - as a general rule and in analogy to paperchart: anything plotted on the chart must be 'true' - all 'magnetic' values should remain detached from chart view as simple numeric (to be compared against readings of magnetic devices, like compass)

wellenvogel, 2021-05-26:
You will have a setting to decice whether to use cog, hdm, hdt for the boat direction. If the selected one is unavailable there is a fall back to cog.
If you enable it you can have a dashed vector for the boat direction.

Littlechay, 2021-05-27:
> You will have a setting to decice whether to use cog, hdm, hdt for the boat direction. If the selected one is unavailable there is a fall back to cog.
> If you enable it you can have a dashed vector for the boat direction.
Awesome, thanks :)

I can't test right now but will this allow both COG and Heading (whichever is selected from HDM HDT) vectors to show? The difference can be very useful for assessing drift in cross tide situations, or for instrument calibration.

wellenvogel, 2021-05-27:
yes.
[image]

wellenvogel, 2021-06-19:
released with 20210619

closed 2021-06-19 by wellenvogel