wellenvogel/avnav #91: v20201227: formatdecimal parameter not working for DepthDisplay widget

W-Geronius, 2021-01-11:
Display format does not change when trying to change DPT display output

wellenvogel, 2021-01-11:
What exactly did you try there?

W-Geronius, 2021-01-12:
I tried to obtain a 00.0 reading by entering formatdecimal parameter 2,1

wellenvogel, 2021-01-12:
leading zeroes are not shown - just for readability. If you inspect the element you will see, that there is an invisible span in front with the leading zeros.
Furthermore the formatting currently does not work if there is no value being available. 
I guess I can change this part.

wellenvogel, 2021-01-13:
implemented with 8d90f36bfa6eb9017035d8b0c0c285eab66c7760

wellenvogel, 2021-01-16:
fixed with release-20210115 https://www.wellenvogel.net/software/avnav/downloads/release/20210115

closed 2021-01-16 by wellenvogel