wellenvogel/avnav #92: suggestion: Open user defined dashboard upon clicking widget within navpage

W-Geronius, 2021-01-11:
It would be nice to open specific dashboard pages (gpspages 1-5) upon clicking a navpage widget.
At present, you will always be taken to gpspage1. 

Example: 
clicking the wind widget could take to directly to user defined gpspage4, 
where all sorts of wind data might be displayed in detail.
This could be defined within the select widget dialog.

wellenvogel, 2021-01-11:
It will try to route you to a page where the item you clicked is shown. If it is page one - just this one.
Maybe an option for the layout editor to have the ability to choose what should happen on the widget click.

W-Geronius, 2021-01-12:
I see, I haven't yet discovered the 'automatic' feature you describe - adding this option to layout editor would certainly enhance user control however.