wellenvogel/avnav #93: suggestion: ocharts cache variants

W-Geronius, 2021-01-11:
at present you have to choose the elements to be displayed onscreen, so the cache writer will prepare the chartset in advance. 

If license permits, it might be a good idea to have at least 2 sets of cache files prepared - during day, you will exclude any light information to reduce excessive screen cluttering, whereas at night, these informations are crucially important.

In OpenCPN light information can be toggled on and off simply by pressing key 'L'

wellenvogel, 2021-01-11:
Could most probably be done.
But for sure requires a substantial effort.
And additional memory/disk space (beside the doubled preprocessing times).
In any case you can always switch immediately - display will be a bit slower in the beginning (not that much an issue on Pi4...).

W-Geronius, 2021-01-12:
As a workaround, I will try to save both sets of avcache files and rename according to actual requirements.
Recreating cache files each time doesn't seem like a good option to me, esp. as Pi4 hasn't become a viable option in my case so far.

wellenvogel, 2021-01-12:
This will not work.
Every time you change the settings, the cache files will become invalid. The same is valid if charts are changing.
And you have to change the settings - otherwise newly shown tiles will be still in the wrong mode...

Maybe you could just try how slow things are getting. I also have a Pi3 on board - and even after changing the settings it is still usable. You just have to be a bit patient in the beginning.
Things will rapidly become better as it will always pre-render tiles around the area you are just using first.

W-Geronius, 2021-01-12:
In fact it does work if you stop the server and replace x.avcache file as well as matching avnav.conf file.
After restarting server, avnav will nicely use the other cache file without rebuilding it.
I will still further test, but it appears to be a solution. 
Disc space (SD) should not be an issue and it saves you some hours of heavy load on your system

closed 2021-01-16 by wellenvogel

rahra, 2023-10-25:
> 
> In OpenCPN light information can be toggled on and off simply by pressing key 'L'

Displaying all the lights does also make the chart very overfull and hard to read in higher zoom levels (zoomed out). This is independently of day/night.

wellenvogel, 2023-10-25:
Unfortunately with the current implementation you cannot easily toggle any feature without the need of invalidating all caches.
On newer hardware this is most probably not that much of a problem.
Anyway I'm working on a new implementation - so maybe I can adress it for the version.