wellenvogel/avnav #94: v20201227 file/download pages: user files won't download on bonjour browser (Android)

W-Geronius, 2021-01-11:
file/download pages: files are not downloaded (neither via icon nor via download option) in Android bonjourbrowser 1.5
downloads work nicely with other browsers

Q: edit and view icons display on files/download pages:
are these icons supposed to be pushed or are they just there to remind users what may be done when clicking the file?

wellenvogel, 2021-01-11:
Bonjour browser currently cannot download at all.
So this is an issue that could go to bonjour browser.
For the Q: basically to inform the user. The problem is the available space for the icons. Chance for invalid button click is simply to big. Therefore the additional dialog in between.

wellenvogel, 2021-03-26:
Bonjour Browser 109, 110 now has this feature.

closed 2021-03-26 by wellenvogel