wellenvogel/avnav #95: suggestion: true vs. magnetic bearings

W-Geronius, 2021-01-11:
Courses derived from electronic charts routing are supposedly true bearings, 
while magnetic bearings are required to steer by compass.
Determining magnetic deviation (VAR) as well as a user definable magnetic deviation table (DEV) might be items to be put on the wishlist

wellenvogel, 2021-01-11:
Nice idea.
Would finally require some decisions on what to use where.

W-Geronius, 2021-01-12:
While there is a WMM model plugin available in OpenCPN, it might not be clear, which VAR apply to different electronic chartsets. This might require some investigation as well.

W-Geronius, 2021-01-13:
pls disregard my last comment - it's nonsense, as VAR is not applied to charts, but to courses, bearings and headings within geografical regions.

wellenvogel, 2021-01-14:
Could become a plugin to compute the magnetic course.

W-Geronius, 2021-04-18:
Charts always plot 'true' courses, as classic navigation practice implies, all course calculations are to be carried out as 'true'
That's AvNav's current state anyway. No changes required here

however:  
all 'true' courses, headings, bearings should to be converted to 'magnetic' (variation/declination plus deviation) if they are to be monitored by means of magnetic devices - either electronically or by user. Similarily, all incoming data (NMEA0183 HDM) must be converted to 'true' (HDT) so they can be plotted correctly

Variation may currently be neglectable in western Europe, however, in other areas up to 30° may be in place
https://maps.ngdc.noaa.gov/viewers/historical_declination/index.html

Deviation too can heavily impact compass readings, mainly if talking of steel ships. Each magnetic device aboard should have its own deviation table attachable

So:
i.e for a helmsman to comfortably steer a calculated 'true' bearing on a magnetic compass, let's say in Canada, it should preferrably be converted to and displayed as 'magnetic' bearing

To achieve this, WMM could be implemented in AvNav, preferrably as plugin using https://www.ngdc.noaa.gov/geomag/WMM/soft.shtml and updates as WMM.COF

optionally, deviation tables could be implemented to be attached to 
- as well electronic magnetic input devices (like imu sensors) - to calculate true readings (potentially plotted to charts)
- as display widgets - to display 'magnetic' values (for the helmsman)
 
In my opinion issue #88 is just very loosely related to this topic, possibly as far as variation might be considered. So I will elaborate my ideas there if you don't mind

PS further reading:
https://www.cruisersforum.com/forums/f2/true-vs-magnetic-bearing-202066.html
although more computerized systems are in place, similar arguments can be discovered in aviation discussions

kdschmidt1, 2021-05-26:
Hello all,
by installing the avnav-more-nmea-plugin (https://github.com/kdschmidt1/avnav-more-nmea-plugin) all necessary data (except for deviation) are available in the avnav store and can be displayed by configuring "default widget" displays in the Layout Editor!
Deviation Tables can be added to avnav-more-nmea-plugin on demand, please leave a message in https://github.com/kdschmidt1/avnav-more-nmea-plugin/issues.
regards
kd

W-Geronius, 2021-05-26:
Yes @kdschmidt1, I am aware of your great plugin. Thank you! I haven't had time so far for a proper test, just saw HDG and HDM working nicely - if incoming data are 'clean' - which they aren't yet  thanks to moitessier hat ;)
As a consequence this issue may be closed, just #88 (orientation of ship on map) remains to be discussed

closed 2021-06-19 by wellenvogel