wellenvogel/avnav #96: v20201227 Dim Button missing in Android Bonjour

W-Geronius, 2021-01-11:
Dim button should be present on navpage in either avnav android or bonjour browser.
In practice however it is absent when using bonjour browser (android)

wellenvogel, 2021-01-11:
Which version of BonjourBrowser?
Did you enable the dim screen in the settings of BonjourBrowser?

W-Geronius, 2021-01-12:
Enabling 'Dim js. Api' in BB v. 1.5 did the trick, thanks. 

Just another Q: 
With Dim Api enabled, display will set either to max. brightness or dimmed state. 
Manual dimming by slider is not functional then. 
Is that standard system behaviour?

wellenvogel, 2021-01-12:
Yes I know.
I already corrected this in AvNav but still not inside Bonjour Browser.
So we can follow this at https://github.com/wellenvogel/BonjourBrowser/issues/4

closed 2021-01-12 by wellenvogel