wellenvogel/avnav #97: v20201227 checksum errors in avnav.log

W-Geronius, 2021-01-13:
avnav.log is filling up with checksum errors, apparently originating from windsensor (norbert-walter).
Not sure if standards forbid small letter checksum values, if so I will report to Norbert.
Otherwise the checking routine should be modified to accept small letter checksum values.
SignalK and OpenCPN are not complaining btw

`GpsdFeeder[standalone feed]-ERROR-invalid checksum in $WIVPW,0.00,N,0.00,M*4c , expected 4C 

GpsdFeeder[standalone feed]-ERROR-invalid checksum in $PWINF,0,27.00,D,0.03,D,38.00,M,136.80,K,73.87,N,12,B,*5b, expected 5B
GpsdFeeder[standalone feed]-ERROR-invalid checksum in $WIMWV,9.00,R,2.00,M,A*2b, expected 2B

GpsdFeeder[standalone feed]-ERROR-invalid checksum in $WIVWR,9.00,R,3.89,N,2.00,M,7.20,K*5b, expected 5B

GpsdFeeder[standalone feed]-ERROR-invalid checksum in $WIVPW,0.00,N,0.00,M*4c, expected 4C

GpsdFeeder[standalone feed]-ERROR-invalid checksum in $WIVPW,0.00,N,0.00,M*4c, expected 4C

GpsdFeeder[standalone feed]-ERROR-invalid checksum in $WIVPW,0.00,N,0.00,M*4c, expected 4C

GpsdFeeder[standalone feed]-ERROR-invalid checksum in $WIVPW,0.00,N,0.00,M*4c, expected 4C

GpsdFeeder[standalone feed]-ERROR-invalid checksum in $WIVPW,0.00,N,0.00,M*4c, expected 4C

GpsdFeeder[standalone feed]-ERROR-invalid checksum in $WIVPW,0.00,N,0.00,M*4c, expected 4C`

free-x, 2021-01-13:
https://github.com/wellenvogel/avnav/blob/97395c7caa79639063955814754045daef04b3da/server/avnav_nmea.py#L249
if sum != valAndSum[1].upper():

norbert-walter, 2021-01-13:
Ok, I check my windsensor firmware. Big letters are also possible.

W-Geronius, 2021-01-13:
I couldn't find any regulations specifying upper- or lowercase hex representation. Possibly both variants are permitted. 
free-x's modification is certainly the smartest fix.

wellenvogel, 2021-01-16:
fixed with release-20210115 https://www.wellenvogel.net/software/avnav/downloads/release/20210115

closed 2021-01-16 by wellenvogel