wellenvogel/avnav #100: signalk-plugin not working correctly

kdschmidt1, 2021-01-17:
With exlusive source from signalk via signalk-plugin the data are added as gps.signalk... but the NMEA status is not changing from red to green and the navigation is not working. if a socketreader on signalk port 10110 is activated and signalk is sending NMEA0183 data the NMEA status switches to green (and remains green even if the port 10110 is not anymore sending data for approx 5 minutes). i assume this is not the correct behavior?

wellenvogel, 2021-01-18:
> With exlusive source from signalk via signalk-plugin the data are added as gps.signalk... but the NMEA status is not changing from red to green and the navigation is not working

Currently AvNav cannot directly use the signalK data for navigation. So this behavior is correct.

> and remains green even if the port 10110 is not anymore sending data for approx 5 minutes

This should not happen. If no NMEA data is received any more the status should switch to red.

wellenvogel, 2022-04-21:
the plugin is not used any more from release 20220421.
So I close this and we should see if the error comes back.

closed 2022-04-21 by wellenvogel