wellenvogel/avnav #102: suggestion: multiple NEMA inputs (Android)

captainfast-git, 2021-01-24:
Maybe you can implement multiple NEMA inputs. For example IP and Bluetooth at the same time. Then it would be possible to get GPS, AIS from IP and temperature, pressure by Bluetooth device.

wellenvogel, 2021-01-24:
sounds possible...
To a certain degree you can already do this - you can use one channel for AIS and the other one for other NMEA data.
But if you get your data via IP:
What is the source? Shouldn't it be possible to collect all data there?

captainfast-git, 2021-01-24:
I have an NEMA wireless device which collects AIS and GPS. This device has no bluetooth. Now I want to add a bluetooth temperature device to my Android tablet. For this it would be nice to be able to select multiple NEMA inputs.

wellenvogel, 2021-04-30:
implemented in 20210424

closed 2021-04-30 by wellenvogel