wellenvogel/avnav #103: v20210115 avnav sound configuration

W-Geronius, 2021-02-01:
default sound configuration is not working, at least when avnav-all is installed on top of OP-Image.
../raspberry/sound.sh is missing - I copied from github and installed mpg123.

I had to alter sound.sh to work with current amixer version:
#amixer set PCM,0 $1
amixer cset numid=3 $1

$1 as sent from avnav appears to hold a percentage, whereas amixer expects a number 0-65353

It's no biggie to change manually but might impose an obstacle to unexperienced users.

W-Geronius, 2021-02-03:
prebuilt OP2 images all contain vlc, which is quite bulky.
https://forum.openmarine.net/showthread.php?tid=3149 (see info files at download site)

instead of installing mpg123, it would be possible to use vlc:
**vlc -Idummy /usr/lib/avnav/sounds/anchorAlarm.mp3 vlc://quit**

wellenvogel, 2021-02-14:
The sound config should be ok with the avnav-raspi package or if you install avnav standalone. But I will add some hint to the description that you need to install mpg123 to make the sound working.
For OP maybe we should open an issue at the integration package:
https://github.com/e-sailing/openplotter-avnav (anyway I already contacted e-sailing about this issue).

wellenvogel, 2021-03-23:
corrected with [20210322](https://www.wellenvogel.net/software/avnav/downloads/release/20210322)

wellenvogel, 2021-03-23:
only for non openplotter...

closed 2021-03-23 by wellenvogel