wellenvogel/avnav #106: GPS parsing errors

mgrouch, 2021-02-10:
2021-02-10 18:32:58,975-7370-[ -1]-GpsdFeeder[standalone feed]-INFO- error parsing nmea data $GPRMC,,V,,,,,,,,,,N*53

Traceback (most recent call last):
File "/usr/lib/avnav/server/avnav_nmea.py", line 297, in parseData
rt['lat']=self.nmeaPosToFloat(darray[3],darray[4])
File "/usr/lib/avnav/server/avnav_nmea.py", line 159, in nmeaPosToFloat
rt=float(grd)+float(min)/60
ValueError: could not convert string to float:

2021-02-10 18:32:58,977-7370-[ -1]-GpsdFeeder[standalone feed]-INFO- error parsing nmea data $GPGGA,,,,,,0,00,99.99,,,,,,*48

Traceback (most recent call last):
File "/usr/lib/avnav/server/avnav_nmea.py", line 265, in parseData
rt['lat']=self.nmeaPosToFloat(darray[2],darray[3])
File "/usr/lib/avnav/server/avnav_nmea.py", line 159, in nmeaPosToFloat
rt=float(grd)+float(min)/60
ValueError: could not convert string to float:

2021-02-10 18:32:58,978-7370-[ -1]-GpsdFeeder[standalone feed]-INFO- error parsing nmea data $GPGLL,,,,,,V,N*64

Traceback (most recent call last):
File "/usr/lib/avnav/server/avnav_nmea.py", line 279, in parseData
rt['lat']=self.nmeaPosToFloat(darray[1],darray[2])
File "/usr/lib/avnav/server/avnav_nmea.py", line 159, in nmeaPosToFloat
rt=float(grd)+float(min)/60
ValueError: could not convert string to float:

2021-02-10 18:32:59,899-7370-[ -1]-GpsdFeeder[standalone feed]-INFO- error parsing nmea data $GPRMC,000000,V,0000.0000,S,00000.0000,W,0.0000,0.000,000000,,*35

Traceback (most recent call last):
File "/usr/lib/avnav/server/avnav_nmea.py", line 304, in parseData
rt['time']=self.formatTime(self.gpsTimeToTime(gpstime, gpsdate))
File "/usr/lib/avnav/server/avnav_nmea.py", line 142, in gpsTimeToTime
date=datetime.date(completeyear,int(gpsdate[2:4]),int(gpsdate[0:2]))
ValueError: month must be in 1..12

2021-02-10 18:32:59,903-7370-[ -1]-GpsdFeeder[standalone feed]-INFO- error parsing nmea data $GNRMC,,V,,,,,,,,,,N*4D

Traceback (most recent call last):
File "/usr/lib/avnav/server/avnav_nmea.py", line 297, in parseData
rt['lat']=self.nmeaPosToFloat(darray[3],darray[4])
File "/usr/lib/avnav/server/avnav_nmea.py", line 159, in nmeaPosToFloat
rt=float(grd)+float(min)/60
ValueError: could not convert string to float:

2021-02-10 18:32:59,905-7370-[ -1]-GpsdFeeder[standalone feed]-INFO- error parsing nmea data $GNGGA,,,,,,0,00,99.99,,,,,,*56

Traceback (most recent call last):
File "/usr/lib/avnav/server/avnav_nmea.py", line 265, in parseData
rt['lat']=self.nmeaPosToFloat(darray[2],darray[3])
File "/usr/lib/avnav/server/avnav_nmea.py", line 159, in nmeaPosToFloat
rt=float(grd)+float(min)/60
ValueError: could not convert string to float:

2021-02-10 18:32:59,907-7370-[ -1]-GpsdFeeder[standalone feed]-INFO- error parsing nmea data $GNGLL,,,,,,V,N*7A

Traceback (most recent call last):
File "/usr/lib/avnav/server/avnav_nmea.py", line 279, in parseData
rt['lat']=self.nmeaPosToFloat(darray[1],darray[2])
File "/usr/lib/avnav/server/avnav_nmea.py", line 159, in nmeaPosToFloat
rt=float(grd)+float(min)/60
ValueError: could not convert string to float:

2021-02-10 18:32:59,973-7370-[ -1]-GpsdFeeder[standalone feed]-INFO- error parsing nmea data $GPRMC,000000,V,0000.0000,S,00000.0000,W,0.0000,0.000,000000,,*35

Traceback (most recent call last):
File "/usr/lib/avnav/server/avnav_nmea.py", line 304, in parseData
rt['time']=self.formatTime(self.gpsTimeToTime(gpstime, gpsdate))
File "/usr/lib/avnav/server/avnav_nmea.py", line 142, in gpsTimeToTime
date=datetime.date(completeyear,int(gpsdate[2:4]),int(gpsdate[0:2]))
ValueError: month must be in 1..12

2021-02-10 18:32:59,974-7370-[ -1]-GpsdFeeder[standalone feed]-INFO- error parsing nmea data $GPRMC,,V,,,,,,,,,,N*53

wellenvogel, 2021-02-12:
Thanks for reporting.
Basically it does not affect the work - but it fills the log. So I will avoid writing those errors to the log.

wellenvogel, 2021-03-23:
corrected with [20210322](https://www.wellenvogel.net/software/avnav/downloads/release/20210322)

closed 2021-03-23 by wellenvogel