wellenvogel/avnav #115: Chart Conversion and wx

dominig, 2021-03-10:
Unfortunately the use of wx is always a pain with various compatibility.
On OpenSUSE with OpenCPN is installed we end with several version of wx and it seems that the once selected by avnav is not the expected one.
I am not a wx expert but it does not look friendly. As all the rest of the UI is Web based, it would be nice to get chart conversion also done with a WebUI. It might be easier than to fix wx mess.
```
avnav -g
/usr/lib/avnav/gui/avnav_gui.py:210: wxPyDeprecationWarning: Using deprecated class PySimpleApp. Use :class:`App` instead.
  app = wx.PySimpleApp(0)
Traceback (most recent call last):
  File "/usr/lib/avnav/gui/avnav_gui.py", line 222, in <module>
    frame_1 = AvnavGui(None, -1, "")
  File "/usr/lib/avnav/gui/avnav_gui.py", line 43, in __init__
    Avnav.__init__(self, *args, **kwds)
  File "/usr/lib/avnav/gui/avnav_gui_design.py", line 44, in __init__
    self.__do_layout()
  File "/usr/lib/avnav/gui/avnav_gui_design.py", line 85, in __do_layout
    sizer_9.Add(self.outputDir, 4, wx.LEFT | wx.RIGHT | wx.EXPAND | wx.ALIGN_CENTER_VERTICAL, 3)
wx._core.wxAssertionError: C++ assertion "!(flags & (wxALIGN_BOTTOM | wxALIGN_CENTRE_VERTICAL))" failed at ./src/common/sizer.cpp(2153) in DoInsert(): Vertical alignment flags are ignored with wxEXPAND

```

wellenvogel, 2021-03-10:
Completely agree.
As I already wrote - the converter is also integrated into the "normal" avnav app. So you can upload e.g. kap files via the WebUI and let the converter handle them - nothing special has to be done.
Only drawback for now: You have to upload file by file.
Maybe I will have a look at the wx stuff - but most probably I will more focus on better user experience in the WebUI and get rid of the standalone converter completely.

wellenvogel, 2021-03-28:
Anyway at least on ubuntu/debian it still works using python3-wxgtk4.0.

closed 2021-03-28 by wellenvogel