wellenvogel/avnav #116: AvNav disk io activity

mgrouch, 2021-03-10:
I've noticed that when I have avnav running disk io light stays lighted up all time.
I've run iotop 
and avnav is the most active process (with both reads and writes)

This should be reduced.

Thanks,
--MG

wellenvogel, 2021-03-11:
Would be good to know which version you are using. In some versions there is a lot of unneccessary log writing.

mgrouch, 2021-03-11:
I use latest

wellenvogel, 2021-03-11:
latest release (20210115) or latest daily?

mgrouch, 2021-03-11:
Latest release

wellenvogel, 2021-03-11:
Maybe you could re-check with the latest daily...

wellenvogel, 2021-03-13:
One more question:
Are you using o-charts?

mgrouch, 2021-03-13:
Not me personally but some of my users do.

wellenvogel, 2021-03-13:
o-charts could cause relatively heavy disk activity until the caches are filled up.
But if you are not using it....
I would bet for the log in the latest stable...

wellenvogel, 2021-03-23:
corrected with [20210322](https://www.wellenvogel.net/software/avnav/downloads/release/20210322)

closed 2021-03-23 by wellenvogel