wellenvogel/avnav #117: Two finger zoom issues

mgrouch, 2021-03-13:
Sometimes browser two finger zoom takes over and zooms out whole screen including toolbar buttons, etc
so they get out of view.
It acts as zooming of everything (not just chart as how it's supposed to be)

wellenvogel, 2021-03-13:
Which browser are you using?
I did not notice this on various mobile devices (mostly chrome based) until now.
But i already noticed this easily in firefox.
Basically it seems to be an issue of the browser - not sure if I can control it from the content of the page (will have a look at this).
For some of my testings with firefox I disabled this zooming for the browser.

mgrouch, 2021-03-13:
Chromium on raspberry pi 4

If you use two finger zoom over 
Avnav toolbars and not over charts
then it just zooms whole page instead of chart

wellenvogel, 2021-03-13:
ok, thanks. 
Will dig a bit into it.
Btw: I found chromium not being the best choice for touch screens as it does not support at-spi2. So there is no way to bring up the on screen keyboard on demand (if you have one that supports this - like onboard).
With firefox it works.
I was playing a bit to have some set up for touch screens - and firefox would have been my choice there for this reason. Currently things are a bit tricky - gtk3 and libinput and Xorg on buster are easily crashing the X server...

mgrouch, 2021-03-13:
In my OS image I just have onboard desktop icon for cases where keyboard doesn’t popup automatically. I configured X server to restart automatically. I did see few crashes but they seemed icon drawing related around edges of the screen. Hopefully next to buster version comes out soon from Debian and things be more touchscreen friendlier and gtk2 will go away replaced fully by gtk3.

wellenvogel, 2021-03-13:
Could you just try to add
`body{
  touch-action: none;
}
`
to your user.css in AvNav?
At least in firefox it seems to work.
If it works, I will include this in the next release. Most probably does not work on Safari - but we will see...

For the crashes: if you have some gtk3 app and open a menu, try to touch any of the actions. App becomes unresponsive. If you close it afterwards typically it will crash the X server (or leave it unresponsive). At least found 2 bugs on the net - one for gtk3 and one for Xorg.
Tested this with a couple of apps...
And I wanted to be able to have the browser fullscreen (at least pseudo fullscreen) and the keyboard coming up like you get this on mobile devices.
The crashes seem to be fixed in bullseye (next debian version) - I made some tests and it seems to work. So currently I will wait...

mgrouch, 2021-03-13:
There is no such crashes in my OS image. And it’s buster. Gtk3 menus do not work on touch. But on long touch (when you let your finger go off) they do work. So there is workaround.

mgrouch, 2021-03-13:
I’ve tried body{ touch-action: none; } 

It didn’t make any difference

wellenvogel, 2021-03-13:
Hmm, strange. OK I will test with chromium. Did you reload after inserting?

mgrouch, 2021-03-13:
I did reload. I even rebooted too

wellenvogel, 2021-03-14:
Very strange.
I now tested this with chromium on a touch device.
Without the entry I exactly get this error, with the entry it is gone.
Could you attach your user.css?
Maybe you could check if it arrived correctly using the developer tools?

wellenvogel, 2021-03-15:
The latest daily has the correction included - maybe you could retry...

wellenvogel, 2021-03-23:
corrected with [20210322](https://www.wellenvogel.net/software/avnav/downloads/release/20210322)

closed 2021-03-23 by wellenvogel