wellenvogel/avnav #135: Measure (divider) tool

Littlechay, 2021-05-24:
I would like to see a tool that could drop a waypoint (or use the own ship position if own icon clicked) then allow a second point to dropped. An info box would show bearing and distance between the two points. Both marks should be movable by click-hold or tap-hold and drag.

Ideally multiple instances of these could exist.

wellenvogel, 2021-06-19:
daily 20210619 has a first version. Simply drop a flag at the current center - and have bearing/distance to map center in the center display widget.
For multiple distances you can easily create a temporary route...

wellenvogel, 2022-02-25:
closed with [release-20220225](https://github.com/wellenvogel/avnav/releases/tag/release-20220225)

closed 2022-02-25 by wellenvogel