wellenvogel/avnav #136: Drop mark and edit to select icon

Littlechay, 2021-05-24:
Perhaps this is two requests. 

1. Drop a mark at the vessel position. This could be a pointer/touch icon that when tapped simply drops a mark at the vessels position; there would also be a keyboard shortcut.
2. Drop a mark at any position on the screen by touch and hold or right click
3. The marks would then be editable by clicking and selecting edit options to change name, comment, icon. etc.

Usage - I envisage that this would be used to mark such points of interest as fishing spots, funny readings on the sounder, sail change...  almost anything.

nicknack2007, 2024-08-12:
Yes IT was great If Somebody can add this Function.

Thanks for the Time of the Team WHO deploy this great system!

voilier-kalymnos, 2025-06-23:
Yes it's a great feature if we can create/edit/show waypoints