wellenvogel/avnav #137: S63 vector chart support

Littlechay, 2021-05-24:
I would really appreciate S63 vector chart support as I currently live in NZ and S63 charts are available for free provided you have a user permit (these are available from O-Charts)

wellenvogel, 2021-05-24:
As we already discussed - very unlikely to be implemented as this would require a renderer for such charts.

quantenschaum, 2024-02-08:
It's old but just to mention it. You could grab tiles from https://wetmaps.co.nz/ and cache them using mapproxy.

Littlechay, 2024-02-09:
Yes, but those are raster charts that are no longer updated.

wellenvogel, 2024-02-19:
Maybe there would chances if we could convert the S63 to s57 charts - then we could go for integrating this with the new [ochartsng](https://www.wellenvogel.net/software/avnav/docs/hints/ochartsng.html?lang=en).
One of the points is that we would need a valid user permit to try this...

free-x, 2025-03-01:
https://o-charts.org/shop/de/oesenc/92-neuseeland.html

NZ ist now available in OeSENC format

wellenvogel, 2025-07-23:
closing this due to no further response

closed 2025-07-23 by wellenvogel