wellenvogel/avnav #140: Some widgets do not have night colours

Littlechay, 2021-05-24:
Some widgets do not have night colours so that various parts of them disappear when night mode is activated. 

Wind graphic is one example, the axis labels in the history widget another.

wellenvogel, 2021-05-24:
Would be helpful to get a list of the failing ones.
Especially widgets coming from plugins often do not have correct night mode.

Littlechay, 2021-05-27:
> Would be helpful to get a list of the failing ones.
> Especially widgets coming from plugins often do not have correct night mode.

Ok, This is what I have noticed on my setup;

1. the XTE graphic disappears completely (black on black)
2. COG compass style graphic - centre of compass card remains white (and bright)
3. COG horizontal bar, the entire pane remains white
4. Wind graphic. The angle display and the pointer remain black (black on black)
5. History widget, the y axis labels remain black. There is no colour inversion for the line graph so if it is a dark colour is becomes difficult to see or disappears.

wellenvogel, 2021-05-27:
Thanks for the list.
(1) - still needs a fix, will do
(2) - still need to have a look although first fix already included (20210526)
(3) - all the gauges now have transparent plate as default (20210526). For existing ones you should change the plate color to fully transparent
(4) - fixed in 20210526
(5) - axis fixed in the newest version (20210525). For the colors: For now you must select one that fit's day and night. Maybe I could add a second color selection for night mode.
What you already can do is to tune the history widgets by css (just putting some code into your user.css):
`
.nightMode .speedHistory svg>g>path {
    stroke: yellow !important;
}
`
I added a cssClass "speedHistory" to my widget (layout editor) - and with this setting the path will get yellow color when night mode is active.
Not a perfect solution - but it could help a bit.
The same way you can tune other settings if you want to...
Unfortunately this will not work for the gauges as they are directly drawn to a canvas and so you cannot use css to tune them.

wellenvogel, 2021-06-19:
corrections with 20210619

closed 2021-06-19 by wellenvogel