wellenvogel/avnav #141: Anchor alarm / anchored mode improvements

Littlechay, 2021-05-24:
I believe that the anchor alarm could be made much more useful with a couple of tweaks. 

1. Make the position of the anchor mark editable/moveable in much the same way as my comments on dropping marks and the measurement tool points. touch-hold and drag. 

This would allow the position of the anchor to be marked after the manoeuvre has been completed. 

2. track the vessel position and display the track whilst anchored. display this track in finer distance increments than the vessel track is recorded when under way. The track could be deleted when the anchor position was reset, the vessel gets underway, or on command to avoid cluttering display and memory.

wellenvogel, 2021-08-01:
(1) the dialog for the anchor watch will be improved to give you the option to select the boat pos or the map center as anchor position (with the ability to add a bearing/distance). This should allow (hopefully) to handle all the interesting scenarios.
Additionally I will add the "anchor mode" as parameter for the layout definition on the map page with anchor watch info on bottom left by default.
To get the anchor watch on the nav page just click on the bottom left (waypoint info) to bring up the waypoint buttons.

wellenvogel, 2021-08-01:
First implementation on daily 20210801

Fietsenwiel, 2021-11-09:
I realy would love it when the achoralarm and other alarms also create nmea sentences.
Than you can decide yourzelf what to do with the alarms. For example send if via signalk and mqtt to an mobiel phone.

wellenvogel, 2021-11-09:
Do you have any ideas which sentences would be useful?
At the end things could get a little complicated as alarms have a "state" - and you would have to ensure that "the other end" would get to know about changes in the state. This would require some reliable connection (what NMEA0183 does not provide).
On my todo list for the improvements in the SignalK integration is the integration of alarms.
If someone would be interested I could also extend the plugin api with alarm information. Sou you could easily write a small plugin to do whatever you want...

Fietsenwiel, 2021-11-09:
Maybe nmea is not the way to go but bidirectional signalk sentices are....
Like: 
vessels.self.notifications.navigation.anchor.currentRadius (you can test and take action from signalk plugins)
vessels.[uuid].notifications..notifications.mob.*

And when listening to this signalk alarms a alarm can also made from an other source and being shown in Avnav.

I am not an expert in this but I think this could help bringing an alarm to external devices....

Fietsenwiel, 2021-11-09:
I now found out that it is possible to use the AVNCommandHandler to start a script  when a alarm occurs. In that script I can send somthing to Signalk. Maybe that is a way to get a alarm in signalk.
But receiving an alarm in Avnav would also be desirable when use another plotter besides Avnav.

Fietsenwiel, 2021-11-11:
To get anchoralarms on my pnoe I did the following:

<AVNCommandHandler>
	<Command command="sudo shutdown -P" name="shutdown"/>
	<Command command="/usr/lib/python3/dist-packages/openplotterAvnav/data/sound.sh 90%" name="sound" repeat="1"/>
        <Command command="/usr/bin/python" name="pythonalarm"/>
</AVNCommandHandler>

<AVNAlarmHandler>
  <Alarm name="anchor" command="pythonalarm" parameter="/home/pi/scripts/anchoralarm.py" repeat="1"/>
</AVNAlarmHandler>

The anchoralarm.py script sends something to my phone by Pushbullet but other ways are also possible

wellenvogel, 2021-11-13:
Great solution...
What I would suggest:
Just change your "pythonalarm" command to call your script directly:
`Command command="/usr/bin/python /home/pi/scripts/anchoralarm.py" name="pythonalarm"`
and let the alarm be configured like:
`Alarm name="anchor" command="pythonalarm" parameter="$BASEDIR/../sounds/anchorAlarm.mp3" repeat="1"`
This will still alllow you to have the alarm sound being played in a connected browser. Currently the "parameter" in the alarm definition is expected to contain the sound file...
If your script anchoralarm.py would start with a "shebang" line like
`#! /usr/bin/env python`
and made executable you would not need the /usr/bin/python in front of it.

And in any case I would suggest to have something that will create a sound without any mobile device being connected. Not sure what your anchoralarm use case is - but the one I have in my mind:
I'm on anchor, switching of all the stuff I don't need (tablet, LTE router,...) and still would like to be woken up if the anchor alarm is activated.

wellenvogel, 2022-02-25:
closed with [release-20220225](https://github.com/wellenvogel/avnav/releases/tag/release-20220225)

closed 2022-02-25 by wellenvogel