wellenvogel/avnav #143: v20210527: clicking 'restart' in Server Status page doesn't necessarily restart

W-Geronius, 2021-05-31:
A message 'restart triggered' is displayed, however server doesn't restart if  daemon-reload is required

wellenvogel, 2021-05-31:
Are you sure about the restart?
The daemon-reload is just a hint and does not prevent a restart.

W-Geronius, 2021-05-31:
apparently it didn't restart as the more_nmea plugin was still old version. I am 99,9% sure I clicked 'restart' after installing.
From the the command line, I always had the impression that daemon-reload was a mandatory thing to do ... oh my.

wellenvogel, 2021-06-19:
I cannot reproduce this at all. On my side it always restarts - even with the notification in place.

W-Geronius, 2021-07-03:
In fact I noticed that again since my first report. I will retest using latest version and document properly. Maybe it correlates to the OP2 environment?

wellenvogel, 2021-07-03:
Op2? Hmmm, not impossible but not very likely.

W-Geronius, 2021-07-08:
It has taken some days to reconfirm, sorry for that. On my system, the 'server Status'-page button effectively does not trigger a server restart, see avnav.log below. Command 'sudo systemctl restart avnav' does the job and so does button 'restart AvNav' via AvNav Updater 

`_version 20210618   avnav.log:

...
2021-07-08 10:36:43,180-668-1949-signalk-websocket-INFO-STATUS[builtin-signalk] NMEA, connected at ws://localhost:3000/signalk/v1/stream
2021-07-08 10:36:43,731-668-884-AVNAvahi DBUS loop-INFO-detected new service _nmea-0183._tcp.BOP2010 at 3.0
2021-07-08 10:36:43,732-668-884-AVNAvahi DBUS loop-INFO-detected new service _nmea-0183._tcp.BOP2010 at 4.0

**--> 'restart triggered' via Button in AvNav Server Status Page**

2021-07-08 10:40:12,636-668-668-MainThread-INFO-stopping

**--> no further messages, AvNav continues running**

2021-07-08 10:42:36,695-668-1553-HTTPHandler-INFO-STATUS[0] RUNNING, enabled 0 connections

**--> browser client leaving and reentering AvNav** 

2021-07-08 10:43:39,386-668-2718-HTTPHandler-INFO-added channel connection for 0
2021-07-08 10:43:39,387-668-2718-HTTPHandler-INFO-STATUS[0] NMEA, enabled 1 connections
..._`

wellenvogel, 2021-07-08:
I see...
Ok, I think I already have an idea.
It's most probably the DBUS code with an own main loop that prevents the shutdown.

wellenvogel, 2021-07-11:
first correction in daily 20210711.

W-Geronius, 2022-02-13:
finally verified fix. sorry for the delay!

closed 2022-02-13 by W-Geronius