wellenvogel/avnav #146: Chart sorting and zooming at main-level

PietKnot, 2021-06-21:
1. after starting AvNav and entering the main-level, all charts (incl. additional chart directory about 20 for me) are presented unsorted. Would be nice to have them sorted by name (auto or by user).

2. after selecting a chart for making routes or chart-view most of the time the following msg is presented "Position outside map, center to map now?" and "Ok", the world map (with very small red rectangle) is presented instead of the selected map. Extreme zooming (5-6 times with gesture) or pressing one time the zoom-button is required to view the selected map. 

- why not show the selected map directly even without the center-warning msg? I don't see a real disadvantage

Note: when the center-warning msg is replied with "cancel" you are lost somewhere without a chart thus a reason more to reconsider this behaviour.

wellenvogel, 2021-06-22:
Very good ideas.
Will have a look...

PietKnot, 2021-07-28:
the sorting remark was for the Android version. On Windows it seems a bit better (at least charts-names starting with digits)

wellenvogel, 2021-07-29:
Will just implement a sorting at the browser side. Maybe with an option to select name/last modified.
And the "center to map" can be removed - I also noticed that I agree nearly all the time.

wellenvogel, 2022-02-25:
closed with [release-20220225](https://github.com/wellenvogel/avnav/releases/tag/release-20220225)

closed 2022-02-25 by wellenvogel

PietKnot, 2022-02-25:
Thanks very much!