wellenvogel/avnav #148: Allow server request that triggers a client overlay refresh

miggat, 2021-07-07:
When overlays are modified from outside AvNav (i.e. a service that gather markers from other software and merge them into an AvNav overlay), the web clients aren't notified, and therefore the user needs to refresh the browser (F5/Ctrl+R) in order to see the changes.

A solution could be to export an enpoint on the server where external apps can request to trigger a refresh on the clients.

wellenvogel, 2021-07-07:
I will think about a solution. Maybe an easy approach would be a periodic check of the overlay file ( as similar functiins are already available).
Once the server detects a change it could notify the client.

miggat, 2021-07-08:
That would be a good solution. It's always better IMO to reduce interaction whith other software if the same goal can be achieved without it.

Even better: instead of doing a periodic check, using a watchdog to receive the event when any of the files are modified.

wellenvogel, 2021-07-08:
Agree on the watchdog idea. But will potentially not work on all file systems.
And will require new implementations.
I will have a look.

wellenvogel, 2022-02-25:
closed with [release-20220225](https://github.com/wellenvogel/avnav/releases/tag/release-20220225)

closed 2022-02-25 by wellenvogel