wellenvogel/avnav #149: Customization for AIS Information display

Hans-JR, 2021-07-12:
Currently only the MMSI or ship name is being displayed on the map.
It may be useful to add the possibility to select the displayed information like the speed of the AIS target.
[image]

wellenvogel, 2021-07-15:
Nice Idea. The problem will be to select a nice layout if you want multiple information...

Hans-JR, 2021-07-17:
Well, for sure you cannot have too many data on the map. Like on the attached example for me it is sufficient to display just the name (above the target) and the speed (below the target). These two information in conjunction with the direction of the course vector can help a lot identify the target and estimate any conflicts.

wellenvogel, 2022-02-25:
closed with [release-20220225](https://github.com/wellenvogel/avnav/releases/tag/release-20220225)

closed 2022-02-25 by wellenvogel