wellenvogel/avnav #150: docs/website: link OSM mbtiles repository

bk138, 2021-07-20:
Hi Andreas,
I've got a request for the avnav documentation/website. Couldn't find a website project so am putting this here, hope it's OK:

The [avnav website section about chart formats](https://www.wellenvogel.net/software/avnav/docs/charts.html#chartformats) links to https://wiki.openstreetmap.org/wiki/DE:Locus#https://wiki.openstreetmap.org/wiki/DE:Locus#Offline-Karten regarding downloads of mbtiles charts. However, this wiki page does not list all the mbtiles charts available directly at https://ftp.gwdg.de/pub/misc/openstreetmap/openseamap/charts/mbtiles/

I therefore suggest to link to https://ftp.gwdg.de/pub/misc/openstreetmap/openseamap/charts/mbtiles/ from https://www.wellenvogel.net/software/avnav/docs/charts.html#chartformats directly.

It would help with getting offline charts for avnav more quickly.

wellenvogel, 2021-07-20:
will change this.

wellenvogel, 2021-08-01:
done

closed 2021-08-01 by wellenvogel