wellenvogel/avnav #153: Overlay dialog cannot be used on mobile

wellenvogel, 2021-07-27:
see [https://www.cruisersforum.com/forums/f121/avnav-navigation-in-the-browser-235253-6.html](url)
Meanwhile, a new problem has occurred. I updated to the newest version (june 17th I think) today, and I tried to make a new layer with buoys (nautin). I succeeded in uploading the layer's gpx file. When I wanted to add tgis to a layer, however, it was impossible to edit any of the layer characteristics, or save the layer, because something goes wrong in the browser: when I touch the field that i want to change (minimal zom level) the on-screen keyboard appears briefly, then the layer properties screen changes form (more lines become visible) and the keyboard disappears. I can't enter a value. I tried on an Android tablet and phone.

wellenvogel, 2021-07-27:
can reproduce this

PietKnot, 2021-07-28:
interesting: Nautin is Dutch / Netherlands. Kukel is a name not active at the dutch sailersforum . . . .  maybe i'll contact him.

wellenvogel, 2022-02-25:
closed with [release-20220225](https://github.com/wellenvogel/avnav/releases/tag/release-20220225)

closed 2022-02-25 by wellenvogel