wellenvogel/avnav #161: After august updates avnav fails to start in BBN lysmarine OS

mgrouch, 2021-08-28:
After august updates avnav fails to start in BBN lysmarine OS

There is no clear indication of what fails:

user@lysmarine:/var/log $ systemctl status avnav
\u25cf avnav.service - NMEA multiplexer and Browser based navigation
   Loaded: loaded (/lib/systemd/system/avnav.service; enabled; vendor preset: enabled)
  Drop-In: /usr/lib/systemd/system/avnav.service.d
           \u2514\u2500lys-avnav.conf
   Active: failed (Result: exit-code) since Sat 2021-08-28 04:53:10 BST; 5min ago
  Process: 3023 ExecStartPre=/usr/bin/sh -c chown -R user:user /home/user/avnav/data || true (code=exited, status=0/SUCCESS)
  Process: 3025 ExecStart=/usr/bin/avnav -q -b /home/user/avnav/data -t /usr/lib/avnav/lysmarine/avnav_server_lysmarine.xml (code=exited, status=1/FAILURE)
 Main PID: 3025 (code=exited, status=1/FAILURE)

Aug 28 04:53:10 lysmarine systemd[1]: avnav.service: Service RestartSec=100ms expired, scheduling restart.
Aug 28 04:53:10 lysmarine systemd[1]: avnav.service: Scheduled restart job, restart counter is at 5.
Aug 28 04:53:10 lysmarine systemd[1]: Stopped NMEA multiplexer and Browser based navigation.
Aug 28 04:53:10 lysmarine systemd[1]: avnav.service: Start request repeated too quickly.
Aug 28 04:53:10 lysmarine systemd[1]: avnav.service: Failed with result 'exit-code'.
Aug 28 04:53:10 lysmarine systemd[1]: Failed to start NMEA multiplexer and Browser based navigation.

mgrouch, 2021-08-28:
More info

user@lysmarine:/var/log $ /usr/bin/avnav  -b /home/user/avnav/data -t /usr/lib/avnav/lysmarine/avnav_server_lysmarine.xml
chartdir=/home/user/avnav/data/charts, basedir=/home/user/avnav/data, config=/home/user/avnav/data/avnav_server.xml, starting server...
2021-08-28 05:03:23,078-7547-7547-MainThread-INFO-basedir=/usr/lib/avnav/server,datadir=/home/user/avnav/data
2021-08-28 05:03:23,079-7547-7547-MainThread-INFO-reading config /home/user/avnav/data/avnav_server.xml
2021-08-28 05:03:23,086-7547-7547-MainThread-ERROR-error parsing cfg file /home/user/avnav/data/avnav_server.xml : Traceback (most recent call last):
  File "/usr/lib/avnav/server/avnav_manager.py", line 302, in readConfigAndCreateHandlers
    self.parseDomAndCreateHandlers(filename)
  File "/usr/lib/avnav/server/avnav_manager.py", line 328, in parseDomAndCreateHandlers
    self.domObject=parser.parse(filename)
  File "/usr/lib/python3.7/xml/dom/minidom.py", line 1958, in parse
    return expatbuilder.parse(file)
  File "/usr/lib/python3.7/xml/dom/expatbuilder.py", line 911, in parse
    result = builder.parseFile(fp)
  File "/usr/lib/python3.7/xml/dom/expatbuilder.py", line 211, in parseFile
    parser.Parse("", True)
xml.parsers.expat.ExpatError: no element found: line 1, column 0

2021-08-28 05:03:23,086-7547-7547-MainThread-ERROR-unable to parse config file /home/user/avnav/data/avnav_server.xml
user@lysmarine:/var/log $ xml
xmlpatterns           xmlpatternsvalidator  xmlstarlet            
user@lysmarine:/var/log $ xmlstarlet val /home/user/avnav/data/avnav_server.xml
/home/user/avnav/data/avnav_server.xml - invalid

mgrouch, 2021-08-28:
Somehow avnav blanks /home/user/avnav/data/avnav_server.xml
I have removed that file and restarted avnav. And it started working fine

wellenvogel, 2021-08-28:
There is a slight chance for the cfg to get corrupted if e.g. AvNav crashes during an update.
But normally it should fall back to the latest known good in the same diectory.
Can you provide a directory listing of the cfg file directory?
I will check again in the code and try out with an empty cfg.

wellenvogel, 2022-02-25:
closed with [release-20220225](https://github.com/wellenvogel/avnav/releases/tag/release-20220225)

closed 2022-02-25 by wellenvogel