wellenvogel/avnav #162: avnav doesn't show maps exported via signalk resource plugin

mgrouch, 2021-09-01:
avnav doesn't show maps exported via signalk resource plugin

it does allow to choose them on the first screen

and these charts show up fine in signalk (tuktuk or freebaord-sk)

Thanks,

wellenvogel, 2021-09-01:
Would be helpfull to get some more info.
I.e. versions, example map.
Was tested with some mbtiles.

free-x, 2021-09-01:
i can not confirm. it works as expected
[image]
[image]

mgrouch, 2021-09-01:
In BBN OS on first avnav page I pick sk-noaa-online and they
are not displayed in avnav.

However they display fine in Freeboarrd-sk and tuktuk.

Thanks

free-x, 2021-09-01:
AvNav uses only offline (mbtiles) charts from SignalK. For online resources you can use avnav-mapproxy. among others NOAA is also available there. Advantage of Mapproxy is that all tiles are also cached.
[image]
[image]

avnav-mapproxy is much more powerful than signalk-charts.  ;-)

wellenvogel, 2022-01-17:
As there is no further information and no diagnostics data I will close this issue.

closed 2022-01-17 by wellenvogel