wellenvogel/avnav #165: Integration of Sailing Performance Dashboard

mhomeijer69, 2021-09-25:
Sailing Performance Dashboard calculates a lot of performance related parameters: https://www.capolavoro.nl/home

It would be nice these would be supported by avnav.

Custom / homemade nmea output is already supported and could possibly be supported by avnav or an custom nmea plugin in Signal K. I also asked the developers if want to support signal k output.

PietKnot, 2021-09-25:
a welcome to the dutch Zeilersforum members: Baasklusje, MichaelH  . . . .

wellenvogel, 2021-09-25:
Not exactly sure what you expect from the AvNav side.
If it is about decoding some special NMEA sentences - this easily could be achieved by writing some lines of python code as a plugin - like https://github.com/kdschmidt1/avnav-more-nmea-plugin.
So the question is if someone would be able to do this...

wellenvogel, 2022-01-14:
Anything open here?
Will close this now - please feel free to open a new issue if something more is needed.

closed 2022-01-14 by wellenvogel