wellenvogel/avnav #167: Instable connections to Android server

PietKnot, 2021-09-25:
Andreas, while testing connections to Android server (on different Android versions; with and without firewall active) it basically works but reconnections are often problematical / impossible!

In order to make a pinpointed report what's going on, it would be helpful to ADD to the WebServer status the IP-address it's actual listening on in addition to the text "active on port 8080, external access true". Thanks for considering this small addition.

wellenvogel, 2021-09-25:
I guess this should be doable without too big effort. 
In the mean time you could have a look at the page with the qr codes (reachable from the status page) - there you have the addresses.

Could you describe a bit more in detail which connections you refer to? Just http connections from the AvNav client? or NMEA connections to TCP server?

PietKnot, 2021-09-25:
Just HTTP-connections from Windows 10-Edge to Avnav-Android server and Android-Via-browser  to Avnav-Android server using local-lan  (192.168,2,x). No-root firewall on Android-server.

Android server runs on Android 5 or 9 with or without noroot-firewall (causes network interface change using VPN)

closed 2021-09-25 by PietKnot