wellenvogel/avnav #174: Custom gauge not in Nightmode

Fietsenwiel, 2021-12-03:
Maybe I do something wrong.
I made a custom gauge but now, in nightmode, this gauge is not dimmed.

Is this a software "bug" of configuration problem

Thanks!

wellenvogel, 2021-12-04:
It would be helpful to get some more info, i.e. what exactly did you do, which version of AvNav do you use...
For the gauges: There is only limited support for the night mode. As long as you do not assign colors to elements there are different default colors for day and night mode.
If you implemented your gauge in js you can honor the "nightMode" property in your translateFunction and e.g. assign different colors.
See also #140.

Fietsenwiel, 2021-12-04:
I installed version 20210619 on a raspberry pi.

I didn't know of the nightMode property. Can't find anything about it...

This is what I did:

`var RollGauge = {
    /**
     * a unique name to select the widget
     */
    name: "userRollGauge",
    unit: "Graden",
    /**caption: "RPM",*/
    /**
     * the type of the widget we would like to use
     * in this case: a radial gauge
     * alternatively it could be linearGauge for a linear one
     */
    type: "radialGauge",
    /**
     * we again use our rpm formatter
     */
    /**formatter: rpmFormatter,*/
    /**
     * default min value but we allow the user to overwrite this in the editor
     */
    minValue: -50,
    /**
     * default max value but we allow the user to overwrite this in the editor
     */
    maxValue: 50,
    /**
     * to compute some values that depend on the settings the user has choosen in the layout
     * editor we can have a "translation" function
     * it will receive all settings and the value as an input and can do arbitrary translations
     * this function is completely optional - so just only implement this
     * if you really need to
     * as we would like to have just 10 numbers and the user can choose the range - we need to compute this
     * and we also need to compute the highlight area (if any)
     * the output if this function goes to the gauge - so you can also set any additional parameters you need here
     */
    translateFunction: function (props) {
        //we need to compute the "ticks" that should be displayed from
        //minValue and maxValue
        if (props.minValue !== undefined && props.maxValue !== undefined) {
            var inc = Math.floor((props.maxValue - props.minValue) / 10);
            if (inc < 1) inc = 1;
            var majorTicks = [];
            for (var i = Math.round(props.minValue); i <= props.maxValue; i += inc) {
                majorTicks.push(i);
            }
            props.majorTicks = majorTicks;
        }
        //if the user defined a highlight range
        //we set this
        if (props.startHighlight) {
            props.highlights = [
                {from: props.startHighlight, to: props.maxValue || 10000, color: props.colorHighlight},
                {from: props.startLowlight, to: props.startHighlight || 10000, color: props.colorLowlight}
            ];
        }
        else {
            props.highlights = [];
        }
/**        if (props.startlowlight) {
            props.lowlights = [
                {from: props.startLowlight, to: props.startHighlight || 10000, color: props.colorLowlight}
            ];
        }
        else {
            props.lowlights = [];
        }        
*/      
        return props;
    },

    /**
     * some settings to adapt the gauge to our needs
     * refer to https://canvas-gauges.com/documentation/user-guide/configuration
     * if you want to give the user a chance to adapt some, just add a user parameter definition for it - see below
     */
    startAngle: 90,
    ticksAngle: 180,
    valueBox: false,
    minorTicks: 2,
    strokeTicks: true,
    colorPlate: "#fff",
    borderShadowWidth: "0",
    borders: false,
    needleType: "arrow",
    needleWidth: 2,
    needleCircleSize: 5,
    needleCircleOuter: true,
    needleCircleInner: false,
    animationDuration: 1000,
    //animationRule: "linear"
};
/**
 * the definitions of parameters that the user can change
 * in the layout editor (beside the default parameters)
 * each parameter will become part of the data that the widget will receive
 *
 */
var RollGaugeUserParameter = {
    minValue: {type: 'NUMBER', default: -50},
    maxValue: {type: 'NUMBER', default: 50},
    colorHighlight: {type: 'COLOR', default: "rgba(200, 50, 50, .75)"},
    colorLowlight: {type: 'COLOR', default: "rgba(200, 50, 50, .75)"},
    startHighlight: {type: 'NUMBER'},
    startLowlight: {type: 'NUMBER'}
};
/**
 * uncomment the next line to really register the widget
 */
avnav.api.registerWidget(RollGauge,RollGaugeUserParameter);`

I have read #140 but I don't understand how to use that in this custom gauge

wellenvogel, 2021-12-06:
Basically you need to extend the translateFunction (and potentially the widget parameters). In the translate function just assign different colors depending on whether "nightMode" is set as a property or not. Either use a dimmed color value or just add more widget parameters to ask for dedicated night mode colors. For some default colors it depends on whether you set them in the layout editor or not. But you still could change them in the translate function. 
Just have a look at the default translateFunction for the gauges https://github.com/wellenvogel/avnav/blob/a7ed07d01097f56ed633e3499a6a3e703cfe3887/viewer/components/CanvasGauges.jsx#L14
- it does this for some of the default colors.
Most probably it would especially affect colorLowlight and colorHighlight.

closed 2022-01-14 by wellenvogel